Robotics-STAR-Lab/ApexNav #13: Will the code of real-world deployment be released recently?

Jctrp, 2025-09-26:
Hello! This is a wonderful job!
I'm trying to deploy the code to the real environment, but I have many doubts about how to convert the simulation environment configuration to the real-world configuration  (maybe because I'm coming from a different research direction). 
I noticed that the deployment guide is in your schedule, and I hope it can be released soon to give me some guidance 😊.

Jctrp, 2025-09-28:
Hi, thank you for your prompt reply! 

I will keep a close eye on this work all the time :)

Zager-Zhang, 2025-11-30:
We plan to open-source the planning and control algorithms required for real-world deployment around early December, accompanied by some documentation. Thank you for your attention! @Suhao07

closed 2025-12-11 by Jctrp